martin2250/OpenCNCPilot #172: Please bring the warning message window to be on top of all other windows

deHarro, 2023-02-22:
Hi Martin,

I use pcb-gcode.ulp as tool for generating gcode for routing boards.

pcb-gcode.ulp generates tool change code which is not recognized by OPC.
In turn OPC pops up a window with different messages (which could be safely ignored):

[image]

All well for me, but on all my machines I run OPC on (Büro and at the mill), this window is not displayed topmost, so after opening such a file I sit and stare, waiting for OPC to display the traces... until I remember to look for this message window to discard it :(

Can you please try to make this message window topmost?
(I know, win10 is somewhat tenacious concerning that matter, but most other programs have managed this challenge in some way... perhaps you succeed too :)

Thanks!

Harald

martin2250, 2023-02-22:
Hi Harald,

Thanks for pointing this out, I remember being annoyed by it as well sometimes. I'll take a look at it soon.

Cheers!
Martin

deHarro, 2024-09-11:
Hi Martin,
just hacking your code ;)
On the way I stumbeled over this issue (bug) and fixed it (see [here](https://github.com/martin2250/OpenCNCPilot/commit/02feb27d6061a18e9a8b464bcab043dadefc657c))

But I'm too dumb to make a pull request on this. I don't find the button... 😳

Harald

martin2250, 2024-09-17:
oh nice! I'll test / merge it soon, thanks!