martin2250/OpenCNCPilot #176: Problems with etching a new pcb

deHarro, 2023-05-10:
Hi Martin,

you know, I am using OCP for a long time now, my beginners problems should have gone by now :)
Just now I have a somewhat "larger" design, about 115 x 45 mm in size.

I probed and startet etching this layout for several times now, and every time it seems, that the generated hight map does not make the stylus follow the curvature of the copper surface.
Before starting the milling job I merged the etching and the text part of the g-code to reside in one file, so I see, whether there are overlaps.

To get hold on the problem I tried various modifications:
- Orientation on the mill as from EAGLE (longer side along X axis of the mill)
- Rotated clockwise, to make the longer side follow the Y axis
- Etch only the traces (no text merged, so I have the "original" EAGLE output, which never made problems in the past)
- This file in both orientations as mentioned before

The only difference to other jobs is the size of the board and the comparably large offset in Z along the longer axis. 
I have nearly 0,5 mm difference in Z between X/Y zero and the opposite side of the board.

May this "large" warping have influence on the mathematics when superimposing the warping to the g-code?
I don't think so, but what could be the problem here?

Thanks, Harald

martin2250, 2023-05-10:
Hi Harald,

that is very strange indeed... OCP should not have any issues with large boards (I tested a larger board once, though that one had very few and wide traces). The only thing I can think of would be that the tool does not go deep enough when probing (can you check the settings for the maximum probe depth?) but that should result in an error... In the preview, does it look like it's conforming to the surface properly?

If you don't mind, you could send me the gcode and height map (if you saved that)?

Cheers!

martin2250, 2023-05-10:
Hi Harald,

it looks like there are a few places where there is a large difference between neighboring probe points, maybe you'd have better results using a tighter grid setting (I'd say 2 or 3 times as dense along each axis).

(-0.2 to -0.121) means that the lowest GCode point was raised by 0.079mm, which is in the range shown by the Probing dialog (0.066 to 0.45). It just doesn't seem like a GCode point was exactly on top of the lowest probed point. In my eyes, this is fine.

I can take a closer look at the files tomorrow, but to me it seems like everything is working as intended 😉 

Martin

deHarro, 2023-05-10:
I realize one (big?) difference for this design (against all others before).
This design has an origin different from the corner of the board.
The origin lies 4.064 mm right to the corner of the probed area.

deHarro, 2023-05-10:
Ok, I try a denser probing grid.
Originally I started with 5 by 5 mm which was fine in the past for all designs.
I switched to 7x7 mm for the tests, but no difference to 5x5.
Do you really propose probing with 2x2 mm for this board?? That'll take LONG...
Ok, here we go ;)

martin2250, 2023-05-10:
> The origin lies 4.064 mm right to the corner of the probed area.

that doesn't make a difference, as long as everything stays fixed.

> Do you really propose probing with 2x2 mm for this board?? That'll take LONG...
I'm not making any promises 😅  But with such a large difference (~50-60µm per point) along the short axis...

If that still doesn't work, you can always try etching the areas that are too shallow a second time with a small Z offset to salvage the board.

deHarro, 2023-05-10:
> If that still doesn't work, you can always try etching the areas that are too shallow a second time with a small Z offset to salvage the board.

I tried that in the first place, but since the Z offset lies tilted all over the board (the complete right side doesn't get carved through), the already carved lines got too deep on the following run.
Deleting the "good" area from the g-code is not easy with notepad++ 🙄

martin2250, 2023-05-10:
Then you could still lower selected parts of the toolpath by using height maps generated with mathematical functions. I'm not here this evening, otherwise we could to that together.
That said, none of this should be necessary...

deHarro, 2023-05-10:
> (-0.2 to -0.121) means that the lowest GCode point was raised by 0.079mm, which is in the range shown by the Probing dialog (0.066 to 0.45). It just doesn't seem like a GCode point was exactly on top of the lowest probed point. In my eyes, this is fine.

I can't follow...
The difference from 0.45 to 0.066 is 0.384. I think it's mm(?).
How goes this together with the raise by 0.079 mm you mentioned?

deHarro, 2023-05-10:
> that gcode points might be raised by 0.066 to 0.45 mm, depending on where they are

Yes, I agree.
But I thought, the actual depth of the resulting carve is displayed near the board outline in the viewport.
If that is true, I have a resulting depth of -0,2 mm without height map applied and a resulting depth of -0,121 after applying the map. The diff is 0,09 mm, not 0,384, what I expected from the probing result.

Whre is my fault?
... "are raised..."? Not lowered? ... I'm lost 😖

deHarro, 2023-05-10:
... kaum macht man's richtig, geht's!

The culprit was in deed the loose coupling of router, POM, HDF and board.
I fastened all with new tape, made a probing run with 5x5 grid and routed the combined file (etching and texts), eyes closed, fingers crossed. All went well 😀.

In the meantime I realized, that I came across such problems in the past already and tracked it down back then.  I should have known it...
Obviously, I'm getting old...

Thanks for your support!
Harald

closed 2023-05-10 by deHarro

martin2250, 2023-05-11:
... so eine Überraschung! :P

one last try: the height map shows that the surface is at a height of +0.066 to +0.450 mm. The toolpath is always 0.2mm below the surface, so at a height of −0.134 to +0.250 mm. Since there was no toolpath at the lowest spot, the lowest toolpath ended up being at -0.121mm. Does that explanation make more sense?

deHarro, 2023-05-11:
Oh, wow! I think I get it... more or less...  :) Thanks!
And now I know for sure, why you are able to make all that magic with visualizing and turning and rotating and height mapping our designs :-)
Have a good time!
Harald

deHarro, 2023-05-11:
To make the thread round, here the result of my (our ;) effort...

[image]

The only thing to moan about are those tiny marks scattered around the board (2 on the left and right copper patch ,respectively, as well as 2 on the upper GND patch, 1 on the upper 5V patch, ...).
This are center marks to catch the drill when drilling the holes manually. 
They come from the OLED module I found in the net. I should have controlled it better before usage. 
The OLED modules have 4 holes for screwing them to something, but I do not need that, so I ignored it... 
Not so pcb-gcode.ulp. It reconized it as holes and handles them so. To finish the board I had to manually remove the superfluous drills in the gcode.
To be honest, I feared the action and instead changed the library to have circles instead of holes. Easy :)

Harald

martin2250, 2023-05-11:
looks nice :) only some of the vias could use a bit larger annual rings IMO. the thing with all the extra spot drills is a bit unfortunate, but at least you won't have to mill another board because of that ^^

deHarro, 2023-05-11:
Thanks.
I agree, but I refrain from changing all Rs, Cs and so on in the corresponding libraries of EAGLE. They are designed for industrial boards with through hole plating. There those small rings are ok.
The spot drills are uncomplicated, the board will vanish in a housing and this housing will vanish in the basement boiler room... ;)

And yes, I will populate it soon :)