martin2250/OpenCNCPilot #177: button to place viewport centered flat

deHarro, 2023-05-11:
Hi Martin,

I am used to look at my designs from above and centered.
To get this done, I have to use the mouse, and since my mouse is somewhat jumpy when it comes to moving it only some few pixels, I have a good time to get this done (center the design fullscreen and align the viewport along the X and Y axis).

How about a shortcut which produces exactly that with one click?
Just an idea... :)

Harald

martin2250, 2023-05-11:
Hi Harald,

I'll place this into "nice to have" as well, sorry I don't have enough time to dedicate to OCP rn :/

It won't be a proper orthographic projection as you'd see in a proper 3D modelling software though.

Cheers!

deHarro, 2024-09-08:
Updated the functionality to be more crispy in terms of display quality (snap to orthogonal axis). See [here](https://github.com/martin2250/OpenCNCPilot/pull/192).