martin2250/OpenCNCPilot #185: measure distance of two points in the viewport, snap to coordinates

deHarro, 2024-03-24:
Hi Martin!
Hope you are fine :)

Sometimes I need a quick estimation for the distance of two points or lines displayed in the viewport of OCP.
It would be a great help in some situations.

The cherry on the cake would be if the mouse pointer would snap to the nearest line/point and the coordinates were given in the statusline or elsewhere.
This would be extremely helpful if one is searching for a coordinate in the file (after rotating the workpiece) to e.g. delete a superflous hole in the gcode or something like that.

Just an idea ;)
Harald

martin2250, 2024-04-21:
Hi Harald,

thanks for the suggestion. It would sure be handy, but to be honest right now I don't want to spend the time on implementing that when I'm not using my machine these days :(

Thanks for sticking around :)

Cheers, Martin

deHarro, 2024-04-22:
Hi Martin,

I hope, you sometimes get back to coding (at OCP of course ;)

Till then, have a good time!
Harald