martin2250/OpenCNCPilot #186: Received Bad Status ??

mehmetcihanildiri, 2024-04-19:
I'm using Win 11, GRBL 1.1. OpenCNC Pilot v1.5 is very good. First of all, thank you. My problem is this;
After the connection is established and the job starts, I see the error messages in the photo and the process stops. I need to reset the Arduino.
Thanks very much for the help.
 

[image]

martin2250, 2024-04-21:
Hi,
I can see irregularities in every one of those lines. First line: "FS:," there is no value after "FS". Second line: "MPos:0.000,0.0000.000" the second number has two decimal points. Etc.. It looks like individual characters are lost in the serial transmission. What kind of hardware are you using? If it only appears after starting a job, maybe it is interference from unshielded stepper wires or the spindle?
Anyways, from what I can tell this is not an issue with the software, but your hardware.
Cheers, Martin

mehmetcihanildiri, 2024-04-22:
Your kind reply is very valuable my friend. The source of the problem is an old clone Arduino and CH340 chip. I bought a new original Arduino Uno R3. Everything is very good now. Thanks a lot.

martin2250, 2024-04-23:
great to hear, you're welcome

closed 2024-04-23 by martin2250