martin2250/OpenCNCPilot #187: engine speed

mehmetcihanildiri, 2024-05-01:
Hi there and thanks for using OpenCNCPilot,

before you submit this issue please make sure to check the following things:
- you're using the latest version of grbl (1.1f)
- you have the most recent version of the [.NET Framework](https://www.microsoft.com/net/download/dotnet-framework-runtime) (minimum 4.7)

If you still need help, delete this text and describe your problem and I or another member of our small community will help you.

mehmetcihanildiri, 2024-05-01:
Greetings to everyone. I'm new to the CNC hobby. I have a question. What should be the engine speed when making PCB with a 0.1mm engraving tip? Speed ​​120mm/min. Is there a standard for this?
How do you calculate?

closed 2024-05-01 by mehmetcihanildiri

martin2250, 2024-05-02:
Hi Mehmet,

set your spindle rpm as high as it will go and then just test increasing feedrates until the result is no longer good. With 30k rpm at the spindle, I usually milled circuit boards at 400 mm/min.

Cheers