martin2250/OpenCNCPilot #188: isolation job

mehmetcihanildiri, 2024-05-03:
Friends, scraping all the paths for insulation is very annoying and time consuming. I add a 0.1 mm thick path around the areas to be soldered. This way I save time. Does anyone have similar inventions other than solder mask?   :-)

[image]
[image]

martin2250, 2024-05-04:
not sure why the extra boxes around your pads end up in flatcam, maybe you imported an extra gerber layer? you should be fine with just the copper layer.

mehmetcihanildiri, 2024-05-04:
> Pedlerinizin etrafındaki ekstra kutuların neden düz cam'e dönüştüğünden emin değilsiniz, belki de fazladan bir Gerber katmanı içe aktardınız? Sadece bakır tabaka ile iyi olmalısın.

[image]

Thanks

mehmetcihanildiri, 2024-05-04:
> not sure why the extra boxes around your pads end up in flatcam, maybe you imported an extra gerber layer? you should be fine with just the copper layer.

[image]