martin2250/OpenCNCPilot #195: App connected yesterday, today not anymore: Fatal Error in Work Loop....

McNugget6750, 2024-11-23:
Yesterday, the app worked, connected to my machine and I was able to run a bed leveling test without actually cutting anything. So the process worked. I was even able to load a file and see it with applied autolevel features.

However, today, all I get is what's shown in the screenshot below.
[image]

I have confirmed that the machine still works fine in a serial terminal as well as cnc-gcode-controller.

The error message is not very specific and there is nothing shown in the log.txt.

I then attempted to clone and compile the project but there is no information about development environment used and what packages need to be installed.

Since I loved the software so far I would really like to be able to use it. I'm happy to look through the code and add logging output where needed but right now that's not an option due to missing build instructions.

McNugget6750, 2024-11-23:
I'm running GRBL 1.1g on this machine

McNugget6750, 2024-11-23:
Ha, I just got it to compile and start from Visual Studio 2019 Community Edition. Looking at Machine.cs now, where the main loop appears to live.

McNugget6750, 2024-11-23:
[image]

McNugget6750, 2024-11-23:
The error appears to happen because

`writer.Write("\n$G\n");
writer.Write("\n$#\n");
writer.Flush();`

is sent at the beginning of the work() thread and therefore confirmed by the machine with ok but the sent commands were not added to the sent buffer. Which is why in line 412ff the system throws an error that it received ok without anything in the sent buffer. How this could have changed from yesterday to today is a mystery.

`					if (line == "ok")
					{
						if (Sent.Count != 0)
						{
							BufferState -= ((string)Sent.Dequeue()).Length + 1;
						}
						else
						{
							Console.WriteLine("Received OK without anything in the Sent Buffer");
							BufferState = 0;
						}
					}`

McNugget6750, 2024-11-23:
This happens because my machine uses a start-up command to initialize things I need from the machine. This is a known and established feature of GRBL but I understand not everyone is using this.

This is what my machine spits out when sending the first two commands mentioned above from line 267:

[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]<\r><\n>
ok<\r><\n>
ok<\r><\n>

_and_

[G54:0.000,0.000,0.000]<\r><\n>
[G55:0.000,0.000,0.000]<\r><\n>
[G56:0.000,0.000,0.000]<\r><\n>
[G57:0.000,0.000,0.000]<\r><\n>
[G58:0.000,0.000,0.000]<\r><\n>
[G59:0.000,0.000,0.000]<\r><\n>
[G28:0.750,-109.100,-0.500]<\r><\n>
[G30:0.000,0.000,0.000]<\r><\n>
[G92:-161.650,-109.100,0.000]<\r><\n>
[TLO:0.000]<\r><\n>
[PRB:0.000,0.000,0.000:0]<\r><\n>
ok<\r><\n>
ok<\r><\n>

McNugget6750, 2024-11-24:
As silly as it seems... it was a Windows update that killed the CH340 driver. Reinstalling it solve this issue. That was solid 6h of my life trying to debug is ghost.

https://learn.sparkfun.com/tutorials/how-to-install-ch340-drivers/windows-710

McNugget6750, 2024-11-24:
one last comment: I did replace some init gcodes to make my machine come up more reliably:

				//writer.Write("\n$G\n");
				//Sent.Enqueue("$G\n");
				//writer.Write("\n$#\n");
				//Sent.Enqueue("$#\n");
				writer.Write("$#\n");
				//Sent.Enqueue("\n$#\n");
				writer.Write("$?\n");
				//Sent.Enqueue("\n$?\n");
				writer.Flush();

martin2250, 2024-11-30:
Oh wow, what a journey 😅 Thanks for immediately jumping in with debugging! I'll look into the init code.

Cheers!

McNugget6750, 2024-11-30:
While I was at it, I implemented skew correction as an optional feature. I need for fork the code and create a branch to send you a pull-request as far as I understand git. Maybe I'll do that later.

Regarding the init procedure: I noticed that every time I reboot my PC, I can't connect to GRBL with this OpenCNCPilot. I have to open a serial terminal first, send a command like ? or $$, wait for the response, disconnect, and then connect using your app. I wonder if there is more going on here that should be addressed. I see the send buffer filling up in the UI but it's apparently either not being sent or no 'ok' is received.

deHarro, 2024-11-30:
Hi Timo!

I have two OCPs running here at home (one testbed at the office PC, another for digging material in the hobby room) and neither of it is showing this behavior. Both PC run Win 10 pro, always with latest patch level.
So, from my point of view there should be no problems in connecting to GRBL.

Check the output in the UI for

[image]

directly after you hit "Connect".

As written there, sending "$X" to GRBL should release the "Alarm" state and communication should start.

Clicking the red "Alarm" Button does exactly that for you and GRBL should start communicating.
[image]

The button should turn from "Alarm" to "Idle" and you're done.
[image]

If that doesn't succeed, you can activate "Show status messages" in the Debug pane of OCP and have a look at what's going on.

HTH
Harry

martin2250, 2025-03-23:
Hi @deHarro ,

I don't have a CH340 board here to test this on.

What is exactly the stack trace for the exception [here](https://github.com/martin2250/OpenCNCPilot/blob/master/OpenCNCPilot/Communication/Machine.cs#L479)? Maybe it would be a good idea to have the code from [here](https://github.com/martin2250/OpenCNCPilot/blob/master/OpenCNCPilot/MainWindow.xaml.cs#L117) there as well, to make debugging easier.

If the error is with the lines `writer.Write("\n$G\n"); writer.Write("\n$#\n"); writer.Flush();`, would just adding a delay of 500 ms before help?

Best,
Martin

deHarro, 2025-03-23:
Hi Martin,

I'm just now on setting up a testbed for this. If that is running, I'll test this delay.
I'll keep you posted...

As far as I understood Timo, he suspects missing to add those 3 Write jobs to the sent queue is the culprit.
I tried to implement this, but it did no diference. Or I did it the wrong way (what I suppose :)

Harald

deHarro, 2025-03-23:
I copied OCP to a VM and it performed as expected...
OCP throws the known exception

[image]

At the moment I'm installing VS2019 to are able to debug in the VM. I hope this works...

nota bene:
I do this VM thing because I don't want to de-/reinstall the new/old driver over and over again on my office machine.

deHarro, 2025-03-23:
> Maybe it would be a good idea to have the code from [here](https://github.com/martin2250/OpenCNCPilot/blob/master/OpenCNCPilot/MainWindow.xaml.cs#L117) there as well, to make debugging easier.

I don't get the idea of this...

The link points to line 117 in "MainWindow.xaml.cs".
This code (this routine) is already part of the "normal" code, isn't it?

deHarro, 2025-03-23:
... if you are interested in the output of this routine:

```
Unhandled Exception:
Message:
Nicht genügend Speicher verfügbar, um das Programm weiter auszuführen.
StackTrace:
   bei System.Windows.Media.Composition.DUCE.Channel.SyncFlush()
   bei System.Windows.Interop.HwndTarget.UpdateWindowSettings(Boolean enableRenderTarget, Nullable`1 channelSet)
   bei System.Windows.Interop.HwndTarget.UpdateWindowPos(IntPtr lParam)
   bei System.Windows.Interop.HwndTarget.HandleMessage(WindowMessage msg, IntPtr wparam, IntPtr lparam)
   bei System.Windows.Interop.HwndSource.HwndTargetFilterMessage(IntPtr hwnd, Int32 msg, IntPtr wParam, IntPtr lParam, Boolean& handled)
   bei MS.Win32.HwndWrapper.WndProc(IntPtr hwnd, Int32 msg, IntPtr wParam, IntPtr lParam, Boolean& handled)
   bei MS.Win32.HwndSubclass.DispatcherCallbackOperation(Object o)
   bei System.Windows.Threading.ExceptionWrapper.InternalRealCall(Delegate callback, Object args, Int32 numArgs)
   bei System.Windows.Threading.ExceptionWrapper.TryCatchWhen(Object source, Delegate callback, Object args, Int32 numArgs, Delegate catchHandler)
   bei System.Windows.Threading.Dispatcher.LegacyInvokeImpl(DispatcherPriority priority, TimeSpan timeout, Delegate method, Object args, Int32 numArgs)
   bei MS.Win32.HwndSubclass.SubclassWndProc(IntPtr hwnd, Int32 msg, IntPtr wParam, IntPtr lParam)
ToString():
System.OutOfMemoryException: Nicht genügend Speicher verfügbar, um das Programm weiter auszuführen.
   bei System.Windows.Media.Composition.DUCE.Channel.SyncFlush()
   bei System.Windows.Interop.HwndTarget.UpdateWindowSettings(Boolean enableRenderTarget, Nullable`1 channelSet)
   bei System.Windows.Interop.HwndTarget.UpdateWindowPos(IntPtr lParam)
   bei System.Windows.Interop.HwndTarget.HandleMessage(WindowMessage msg, IntPtr wparam, IntPtr lparam)
   bei System.Windows.Interop.HwndSource.HwndTargetFilterMessage(IntPtr hwnd, Int32 msg, IntPtr wParam, IntPtr lParam, Boolean& handled)
   bei MS.Win32.HwndWrapper.WndProc(IntPtr hwnd, Int32 msg, IntPtr wParam, IntPtr lParam, Boolean& handled)
   bei MS.Win32.HwndSubclass.DispatcherCallbackOperation(Object o)
   bei System.Windows.Threading.ExceptionWrapper.InternalRealCall(Delegate callback, Object args, Int32 numArgs)
   bei System.Windows.Threading.ExceptionWrapper.TryCatchWhen(Object source, Delegate callback, Object args, Int32 numArgs, Delegate catchHandler)
   bei System.Windows.Threading.Dispatcher.LegacyInvokeImpl(DispatcherPriority priority, TimeSpan timeout, Delegate method, Object args, Int32 numArgs)
   bei MS.Win32.HwndSubclass.SubclassWndProc(IntPtr hwnd, Int32 msg, IntPtr wParam, IntPtr lParam)
```

deHarro, 2025-03-23:
Hi @martin2250 !

My VM is up and running, I have attached an Arduino Nano without any periphery.
OCP is starting normally, connecting throws the know exception.

> just adding a delay of 500 ms

Was ist nochmal schnell "delay" in C#? 😕    Ah, ok: Thread.Sleep(500);

I tried the delay before and after the calls to the 3 Writes.
Both makes no difference, still not connecting and throwing the exception.

martin2250, 2025-03-23:
> This code (this routine) is already part of the "normal" code, isn't it?

Yes, but where it currently is, it only runs when OpenCNCPilot is about to crash completely

> This trace (in file "OpenCNCPilot_Crash_Log.txt") is not always written when the exception occurs. Why?This trace (in file "OpenCNCPilot_Crash_Log.txt") is not always written when the exception occurs. Why?

So this happens after copying the code to `Machine.cs`? In theory, it should always write that file when the message shows "Fatal Error in Work Loop". Otherwise there is something fishy going on. Maybe the file immediately gets overwritten or is it missing completely? Maybe `File.AppendAllText` would be more appropriate...

Anyways, the stack trace right now doesn't tell us where the exception occured in the OCP code. Maybe using the debugger and removing the whole try/catch block (lines 244 and 477-481) would tell us more?

deHarro, 2025-03-23:
Shit!
After having closed the VM and reopening it, OCP ("Debug" version) crashes with "access violation". No chance to akt on the program (clicking buttons or something).

Starting the "Release" EXE from today in the morning (when both, release and debug, worked fine).
If I leave OCP alone (mouse not moved over the OCP window) it stays unhampered even for several seconds/minutes.
Moving the mouse into the OCP window, touching any of the active (not disabled) controls (Connect, Settings, Edit,..), OCP collapses.
At the moment the mouse touches any of the controls, the donut comes up, then plopp - OCP is vanished.

Any idea?

deHarro, 2025-03-23:
> after copying the code to `Machine.cs`? 
I didn't copy anything up til now. I work with my private version of OCP, which is a copy of your 1.5.13.0 with some addons

> In theory, it should always write that file when the message shows "Fatal Error in Work Loop".
Where is that file be written, if it is? I found it two times in the folder "bin\debug".
Don't know "when" it was written. As of the files time stamp it must have been written at the first time I started OCP.
But after having deleted the file it was not restored again.
Has the file to be existing?... No, doesn't change anything.

But I currently cannot interact with OCP in the VM.

deHarro, 2025-03-23:
Update
To check with bCNC I had to enable "3D acceleration" within the VM.
This seems to kill OCP.
I disabled the 3D acceleration, now I'm back in he game :) (OCP starts and I can set the COM port et al).

deHarro, 2025-03-23:
> So this happens after copying the code to `Machine.cs`?

Do I get you right?
I have to copy this code from line 117 in "MainWindow.xaml.cs" into "Machine.cs"?
Plainly that?
If so, where to in "Machine.cs"?

deHarro, 2025-03-23:
If I click "connect" I get this in the Console window of VS2019:

Ausnahme ausgelöst: "System.AggregateException" in mscorlib.dll
Der Thread 0xbe0 hat mit Code 0 (0x0) geendet.
Der Thread 0x1048 hat mit Code 0 (0x0) geendet.

deHarro, 2025-03-23:
@martin2250 
Would it be easier and more direct to switch to mail communication for that debug thing?

deHarro, 2025-03-23:
I enabled "stop on all exceptions" and got this when clicked on "connect":

[image]

deHarro, 2025-03-23:
System.AggregateException contains exactly 1 inner message:
"Der E/A-Vorgang wurde wegen eines Threadendes oder einer Anwendungsanforderung abgebrochen."

I caught that by replacing your code in "Machine.cs, at end of module "Work()", line 477ff
```
catch (Exception ex)
{
	RaiseEvent(ReportError, $"Fatal Error in Work Loop: {ex.Message}");
	RaiseEvent(() => Disconnect());
}
```

with

```
catch (AggregateException ex)
{
	foreach (var innerEx in ex.InnerExceptions)
	{
		Console.WriteLine(innerEx.Message);
	}
}

```

deHarro, 2025-03-24:
I set up two parallel debug sessions. 
1.) One with the old CH340 driver which lets OCP connect to GRBL.
2.) One with the new CH340 driver which throws the exception on connect.

Single stepping in parallel I reached a point when 1.) gets the response 
**Id = 73803, Status = RanToCompletion, Method = "{null}", Result = "<Idle|MPos:0.000,0.000,0.000|FS:0,0>"** 
from GRBL, whereas 2.) gets the response 
**Id = 7122, Status = Faulted, Method = "{null}", Result = "{Noch nicht berechnet}"**

I made some screen dumps and now stop staggering around.
@martin2250 has to step in...

Harald

martin2250, 2025-03-26:
Hi Harald,

thank you for putting the time in to debug. To be honest, I don't understand why this is happening. Seems to me like the error occurs within C#'s `async StreamReader.ReadLine()` for some reason. I'll check if maybe I can borrow a CH340 board somewhere, I think this goes too deep for remote debugging...

Until then, the solution with changing the driver seems like a good workaround. If that fixes it, I don't want to put in too much time. I guess that fixing this would require some bigger changes to the serial-receive code.

Best,
Martin

McNugget6750, 2025-03-27:
I have to replace my driver after any windows update and roll it back to the 2014 version for any of my CH340 to work. It's insane. I can't even program an Arduino using the newer drivers. I wonder if this is a scheme to push them off the market at this point as they are ubiquitous and everywhere

deHarro, 2025-03-27:
Hi Timo!

This I cannot confirm.
I have loads of Arduino Nanos here at my office computer (and at other computers here) and all are communicating well with the new (and the old) CH340 driver. No issues with programming or whatsoever.
And we (Guru and I) tested several programs in conjunction with GRBL, all working normally with the new driver.

The only program not working with the new (up-to-date) driver is OCP.

Just my 2 cents...

Harald

McNugget6750, 2025-03-27:
I'll go and check on a different Windows PC. Maybe I am facing a configuration issue on this particular system

deHarro, 2025-03-28:
Hi @martin2250!

> Maybe it would be a good idea to have the code from [here](https://github.com/martin2250/OpenCNCPilot/blob/master/OpenCNCPilot/MainWindow.xaml.cs#L117) there as well, to make debugging easier.

I reread your advices above once again and suddenly I realized what you wanted me to do 😁. (Zarah Leander lässt grüßen ;)

The first 2 lines is what I implemented as crutch for my long term memory 😉.

I adapted the code from "MainWindow.xaml.cs" to "Machine.cs" (starting from "//Exception e = ...") ...

```
	catch (Exception ex)
	{
		//RaiseEvent(ReportError, $"Possibly bad CH340 driver. Look at github, issue #195");	// deHarry, 2025-03-18
		//RaiseEvent(ReportError, $"StackTrace: {ex.StackTrace}");								// deHarry, 2025-03-18
		/* 
		 * StackTrace mit aktuellem CH340 Treiber (Version von 2024-10-16)
		 * "bei System.Threading.Task.Task
		 *  1.GetResultCore(Boolean waitCompletionNotification) bei OpenCNCPilot.Communication.Machine.Work()
		 *  in D:\...\OpenCNCPilot\Communication\Machine.cs:Zeile 408."
		 *  Workaround mit Treiber von 04/2014.
		 */

		//Exception e = (Exception)ea.ExceptionObject;

		string info = "Unhandled Exception:\r\nMessage:\r\n";
		info += ex.Message;
		info += "\r\nStackTrace:\r\n";
		info += ex.StackTrace;
		info += "\r\nToString():\r\n";
		info += ex.ToString();

		MessageBox.Show(info);
		Console.WriteLine(info);

		try
		{
			System.IO.File.WriteAllText("OpenCNCPilot_Crash_Log.txt", info);
		}
		catch { }

		RaiseEvent(ReportError, $"Fatal Error in Work Loop: {ex.Message}");
		RaiseEvent(() => Disconnect());
	}

```

...and got this as output:

```
Ausnahme ausgelöst: "System.AggregateException" in mscorlib.dll
Unhandled Exception:
Message:
Mindestens ein Fehler ist aufgetreten.
StackTrace:
   bei System.Threading.Tasks.Task`1.GetResultCore(Boolean waitCompletionNotification)
   bei OpenCNCPilot.Communication.Machine.Work() in C:\Users\harry\Desktop\OpenCNCPilot\OpenCNCPilot\Communication\Machine.cs:Zeile 408.
ToString():
System.AggregateException: Mindestens ein Fehler ist aufgetreten. ---> System.IO.IOException: Der E/A-Vorgang wurde wegen eines Threadendes oder einer Anwendungsanforderung abgebrochen.
   bei System.IO.Ports.SerialStream.EndRead(IAsyncResult asyncResult)
   bei System.Threading.Tasks.TaskFactory`1.FromAsyncTrimPromise`1.Complete(TInstance thisRef, Func`3 endMethod, IAsyncResult asyncResult, Boolean requiresSynchronization)
--- Ende der Stapelüberwachung vom vorhergehenden Ort, an dem die Ausnahme ausgelöst wurde ---
   bei System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   bei System.IO.StreamReader.<ReadBufferAsync>d__97.MoveNext()
--- Ende der Stapelüberwachung vom vorhergehenden Ort, an dem die Ausnahme ausgelöst wurde ---
   bei System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   bei System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   bei System.IO.StreamReader.<ReadLineAsyncInternal>d__60.MoveNext()
   --- Ende der internen Ausnahmestapelüberwachung ---
   bei System.Threading.Tasks.Task`1.GetResultCore(Boolean waitCompletionNotification)
   bei OpenCNCPilot.Communication.Machine.Work() in C:\Users\harry\Desktop\OpenCNCPilot\OpenCNCPilot\Communication\Machine.cs:Zeile 408.
---> (Interne Ausnahme #0) System.IO.IOException: Der E/A-Vorgang wurde wegen eines Threadendes oder einer Anwendungsanforderung abgebrochen.
   bei System.IO.Ports.SerialStream.EndRead(IAsyncResult asyncResult)
   bei System.Threading.Tasks.TaskFactory`1.FromAsyncTrimPromise`1.Complete(TInstance thisRef, Func`3 endMethod, IAsyncResult asyncResult, Boolean requiresSynchronization)
--- Ende der Stapelüberwachung vom vorhergehenden Ort, an dem die Ausnahme ausgelöst wurde ---
   bei System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   bei System.IO.StreamReader.<ReadBufferAsync>d__97.MoveNext()
--- Ende der Stapelüberwachung vom vorhergehenden Ort, an dem die Ausnahme ausgelöst wurde ---
   bei System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   bei System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   bei System.IO.StreamReader.<ReadLineAsyncInternal>d__60.MoveNext()<---

Der Thread 0x914 hat mit Code 0 (0x0) geendet.
Der Thread 0x410 hat mit Code 0 (0x0) geendet.
```

Perhaps it helps...

Harald

deHarro, 2025-03-28:
Hi Martin,

I did so, connecting is ok, but I get more updates in the below right corner of the viewport:

[image]

This pattern repeats over and over.

Normally I get only this...

[image]

..., the output fades and that's it.

Perhaps I have some remainders of my debug outputs left in the code, I will check that.

Besides that, even carving a board seems to run. Ok, only at the office computer without any hardware (besides the Arduino), but I have a good feeling 😊👍

I'll be back with info, if I found the debug lines...

deHarro, 2025-03-28:
Ok, this additional output is no debug output leftover.
I copied the "original" code from my main (office) machine into the VM and checked for the error without your last commit.
Ok, error occurs.
Then expanding line 408... error is gone, but the additional output is generated again.

In the meantime this output has changed a bit to this:

[image]

martin2250, 2025-03-28:
huh, in your first screenshot it looks like the status gets cut off sometimes -> data is lost. do the coordinates in the top status bar update or are they maybe less responsive than usually?

I'm not really happy with this solution, because important messages might get lost

martin2250, 2025-03-28:
maybe one thing to test would be adding a debug message into the `if` to see how often that error occurs

deHarro, 2025-03-28:
Hi Martin,
I'll do this tomorrow.
Harald

deHarro, 2025-03-29:
Wow!
The error occurs about 150 times / second (after "connect").

```
//int errCounter = 0;

if (lineTask.Status != TaskStatus.RanToCompletion)
{
    DateTime now = DateTime.Now;
	Console.WriteLine("TaskStatus: Not completed {0} - {1}", errCounter, now);
	errCounter++;
	continue;
}
```

Result:
```
TaskStatus: Not completed 70 - 29.03.2025 12:00:16
TaskStatus: Not completed 71 - 29.03.2025 12:00:16
TaskStatus: Not completed 72 - 29.03.2025 12:00:17
TaskStatus: Not completed 73 - 29.03.2025 12:00:17
TaskStatus: Not completed 74 - 29.03.2025 12:00:17
TaskStatus: Not completed 75 - 29.03.2025 12:00:17
...
TaskStatus: Not completed 222 - 29.03.2025 12:00:17
TaskStatus: Not completed 223 - 29.03.2025 12:00:17
TaskStatus: Not completed 224 - 29.03.2025 12:00:17
TaskStatus: Not completed 225 - 29.03.2025 12:00:17
TaskStatus: Not completed 226 - 29.03.2025 12:00:18
TaskStatus: Not completed 227 - 29.03.2025 12:00:18
TaskStatus: Not completed 228 - 29.03.2025 12:00:18
```

Concerning your former question about the responsiveness of the status values...
This is not rated easily, because the debugging runs in the VM and therein the operation is not smooth anyway.
Nevertheless I daresay the update is somewhat laggy or jumpy.

martin2250, 2025-03-30:
mh :/

yes, that's what I thought - it misses "ok"s and never clears the lines from the buffer...

not sure what to do, probably we'd have to move away from the async ReadLine() method.

deHarro, 2025-03-30:
Whatever you think that might help... 😉

Do you know what command goes to GRBL, GRBL answers with this 'ok' "without something in the sent queue"?
Perhaps this is the location to fix the problem.

deHarro, 2025-04-01:
> probably we'd have to move away from the async ReadLine() method.

Hi Martin,

perhaps you might check at [GRBL Plotter](https://github.com/svenhb/GRBL-Plotter), how they deal with the COM port?

Just an idea...
Harald

McNugget6750, 2025-04-01:
the reason is that the initialization happens just by manually sending gcode programmatically. Those codes aren't added to the queue instead. So when grbl sends back "ok" there is nothing to okey in the queue and we're out of sync

McNugget6750, 2025-04-01:
ah: to add to that, in my case my machine has a coordinate system programmed in. Upon boot or homing, these commands are internally executed and (for some reason) externally confirmed with "ok" - as far as I recall.

deHarro, 2025-04-04:
I just checked: 
Commenting out those 3 lines before the while(true) loop does not change the error.
-> So, the error seems not to be in conjunction with those commands.

Additional to that I was wondering about the fact that a **CH340 driver update** can imply such a totally different behavior of the formerly cleanly running program.
This points me to a problem having nothing to do with the communication between OCP and GRBL.

deHarro, 2025-05-03:
I'm currently get starting with laser-cutting and engraving, so I digged into LaserGRBL as GUI for this.
...and found this interesting article about our driver problem:

[LaserGRBL](https://lasergrbl.com/usage/arduino-connection/)

or [directly](https://lasergrbl.com/november-2024-windows-driver-update/)

According to the article and the over there linked MS statement, Microsoft did the same cruel measure as FTDI some years ago, busting all applications using non-original FTDI chips for USB commmunication 😕.
Now non-original CH340 chips are the victims.
But I think it's not Microsofts mistake. They only provide the customers with an updated driver from WCH-IC, the manufacturer of the CH340 chip.

Ok, having read that, we obviously can stop investigating in workarounds, can't we?

The most annoying "solution" I found is this:
https://answers.microsoft.com/en-us/profile/9c710b11-cfb2-4700-b98f-a575621e0cc2

It's located on page 2, 3rd Message, author is Gyozo Szabo:

```
**Solution V2.0, that works for me**

1. Delete all content in `c:\Windows\SoftwareDistribution\Download\Install\` (assuming only CH341 files are present there).

2. Copy all the following files to the above path:
- `ch341SER.CAT` 10466 2015.2.6 6:43.0
- `CH341S64.SYS` 59904 2015.1.26 0:0.0
- `CH341SER.SYS` 41472 2015.1.26 0:0.0
- `CH341SER.INF` 6678 2014.8.7 23:0.0
- `CH341SER.VXD` 20089 2008.12.18 0:0.0
- `CH341S98.SYS` 19680 2007.6.11 23:0.0
- `CH341PT.DLL` 6712 2005.7.29 23:0.0

3. Set all files to "Read Only."

4. In **Device Manager** > **Ports (COM & LPT)**, uninstall the devices and check the box for **Delete the driver software for this device**. Then, scan for hardware changes.

That's ALL!
```

A sound mixture of this rundown and a former installation of the old driver by its installer would yield the same result and finally would rescue a user with recurring problems by automatic driver updates.

...

Only the fact, that bCNC ist still working with the updated driver, is confusing me.

Harald

deHarro, 2025-05-03:
Hi @martin2250 !
In addition to that, I found a programmatical workaround in the code of LaserGRBL:

[https://github.com/arkypita/LaserGRBL/commit/857ab1b2d8d196c2459de77cf5d883599c68eac1](https://github.com/arkypita/LaserGRBL/commit/857ab1b2d8d196c2459de77cf5d883599c68eac1)

  Perhaps you understand what he did to circumvent the driver problem in his application...

And yes, LaserGRBL runs smoothly with the current driver from Microsoft :)

McNugget6750, 2025-05-04:
oh my god what an investigation! Great find! I'll try this!

deHarro, 2025-05-04:
Hi Timo!
May I ask, what of the last two depicted approaches you will try?
1. Writeprotecting the files or
2. Implementing the error handling procedure of LaserGRBL into OCP?

Thanks!

martin2250, 2025-05-11:
Hi @deHarro , that's a good find!

If I understand the fix right, it's simply ignoring errors and sometimes introducing a delay. That's basically the same thing that you tested in this code: https://github.com/martin2250/OpenCNCPilot/issues/195#issuecomment-2763301941

The thing is that this will, without additional changes, lead to the same problem as you encountered: OCP will think that the buffer fills up, because many "ok" messages from grbl will get lost. I'm not sure what LaserGRBL does to keep the buffer level in sync. Anyways, this would probably require some pretty big changes in Machine.cs that I wouldn't want to release without being able to test them thoroughly.

For now, Option 1 seems like the better option as it doesn't require any changes to the (not well written, but) time-proven code.

I'm surprised that even the WCH chips get counterfeited with the prices already that low...

deHarro, 2025-05-11:
Hi @martin2250, yes, I was taken by surprise when I realized, that so many apps are having the same problem as OCP.

I'm not sure, that his fix has to do with missing or superfluous "ok"s  from GRBL.
He's counting CH340 exceptions and from time to time (every 10) he tries: `CloseCom(false);`
What does this call? Especially the "false" parameter?

I just found another peculiarity...
In the meanwhile I flashed another sketch onto the Arduino I used to debug OCP on the VM. This other sketch does NOT communicate with OCP (no GRBL code flashed), but sends some status information and otherwise is quiet on the serial port. 
This leads to the same error message in OCP.

**-> This is hardening my opinion, that the errors are not related to superfluous "ok"s from GRBL.**

And onwards... (spoiler: Not related to our error :)
Using an **Arduino with USB-C** connector with only the OptiBoot bootloader (and therefore not answering nor sending anything), I get no error message, OCP connects and starts querying GRBL, being happy without any answers.

Flashing this Arduino with GRBL V1.1h
...
Done.
The communication is running fluently, no complaints from any of the partners and a gcode file I loaded, is toiled away as usual.

Explanation: This Arduino clone has an unbranded chip with identical case and pattern of a CH340, but it is definitely not a CH340.
I know that, because one of the batch of Arduinos I ordered was faulty on serial comm, so I replaced the USB chip with a CH340 I had on hands... resulting in no comm at all now.
This formerly only acting faulty Arduino is not recognized by Windows any more.
The driver for this unbranded chip is "usbser.sys" originated by Microsoft Corp.
Had I known this beforehand, I had omitted this test ;)