martin2250/OpenCNCPilot #196: Probe first moves X, Y instead of starting at 0, 0

thomergil, 2025-04-26:
Thank you very much for OpenCNCPilot.

When I manually zero, use G10, and then set up a new Probing, and Run it, the CNC moves up, down, and to the lift before probing. I had set the X/Y coordinates to greater-than-zero values, and the red dots in the UI show the correct area? Why is this happening?

thomergil, 2025-04-26:
Aha! Under Manual Probing Probe Offset was set to 10, 10! Setting that to zero solves the problem.

This feels like a bug to me, though. I would think that Probing is unrelated to Manual Probing.

martin2250, 2025-04-28:
Huh, thanks for the report. The idea for that is to accommodate external probes (similar to the BLTouch probe for 3D printers). We want to use that for height maps as well, it's just misnamed.

closed 2025-04-29 by thomergil