martin2250/OpenCNCPilot #198: Thinnest isolation

thomergil, 2025-04-28:
@deHarro, @martin2250, what's the thinnest isolation track you've been able to make? 0.4mm seems doable, but 0.2mm seems pretty much impossible. Does that match your experience?

I've tried a 40º V-bit, anywhere from 0.025mm to 0.04mm depth; the problem seems that it either doesn't go deep enough to carve the copper away, or it goes deep enough but then leaves no trace copper behind in the middle of the isolation track.

What dimensions do you recommend?

Thank you.

deHarro, 2025-04-28:
Hi thomergil!

I once managed to carve 0.2 mm structures, see here: [https://harald-sattler.de/html/rollercontrol_v4.htm](https://harald-sattler.de/html/rollercontrol_v4.htm)
The site is in German, but you can use Google translate (use link at the bottom of each page).

In a nutshell:
40° V-bits are way too flat for carving pcbs.
Use maximum 20° or even 10° for finer structures.
Make test carvings with different depth to get the optimal depth. I use 0.069 mm.
I first startet with 0.04 mm, reconing with the normal copper clad ~30µ, but that was not constructive.

Martin once explained the reason for this deviation. I understood his explanation as so:
During the probing process, GRBL instantly stops when the probe pin is triggered.
When carving, GRBL decelerates slowly to reach the exact depth.
Those two approaches lead to different results, measuring and carving, respectively.

Harald

martin2250, 2025-04-28:
Very good summary. I recommend 30° for general PCBs, but for very fine traces use less. The depth will depend on
- the probing speed (some delay in the motion system, not instantaneous reaction on contact)
- your fixturing (the PCB will deflect during cutting)
- your machine (the bit will deflect during cutting, sometimes you can see that the center is not etched deep enough but the sides are, are the machine is less stiff in the middle of the X-travel)

> Martin once explained the reason for this deviation
the explanation you gave is the one for the difference in the position in the "manual probe" tab and the machine position after a G38.2. The height map uses the contact position and not the end position as the first is more accurate and repeatable.