martin2250/OpenCNCPilot #199: Order of operations? Milling x2, drilling? Or Milling, drilling, milling?

thomergil, 2025-04-28:
@deHarro, @martin2250, I'm curious about the order of operations. I was under the impression that one is supposed to mill, drill, flip over the copper clad, and then mill again.

However, probing a pre-drilled board (that is, after milling, drilling, and flipping) leads to incorrect depth measurements when the mill happens to probe a drilled hole. OpenCNCPilot is showing a green spot where it happened to probe a hole.

Would you recommend mill, flip, mill, flip back, drill? What do you do? Thank you.

deHarro, 2025-04-28:
Hi Tom!

I seldomly do two sided boards. 

This said, you should 
1. Probe side A
2. Etch side A (etching in the sense of carving ;)
3. Drill 2 alignment holes (if required. Perhaps you have other means of adjusting the board?)
4. Flip board
5. Probe side B
6. Etch side B
7. Drill side B (holes for TH elements)
8. Mill side B (perimeter of board)

Harald

thomergil, 2025-04-28:
I start out making alignment holes, but in all other ways, this sounds good. Thank you.

martin2250, 2025-04-28:
Not much to add, I only made a handful of double-sided boards as well. Just my feeling for my particular PCB base material: it doesn't really matter what the order of operations is. If you have issues with pads ripping loose during drilling, consider drilling earlier. But there is, of course, the issue of probing afterwards...

deHarro, 2025-04-29:
> If you have issues with pads ripping loose during drilling,

This is nothing I realized ever 😮. Perhaps related to the board quality? But I use cheapest material from our chinese friends.

I normally do not change the standard ring size around a hole (for TH components). Only exception from this are vias I have to place to solder a wire as replacement for a doublesided design. Here I change the standard copper ring to a somewhat bigger version.

But I use endmills to drill holes, not a normal twist drill. Perhaps this prevents the pads from being ripped off?