martin2250/OpenCNCPilot #200: Pause and Start confusion

thomergil, 2025-04-28:
@deHarro, @martin2250,

There are two Start buttons. One is in the File menu, the other is near the top of the UI. I assume they have different functions, but I don't always understand what it is. Can you clarify?

Semi-related, sometimes the machine randomly pauses in the middle of drilling. I then need to press Start for it to continue. Why?

Thank you.

deHarro, 2025-04-28:
Hi thomergil!

The first question is something, Martin should elaborate on 😉

This pausing on drilling is related to tool changes for different hole diameters (as far as it occurs at my jobs, generated with EAGLE pcb-gcode.ulp).

By the way... are you Thomas Ergil? So may I address you as "Tom"? Thanks!
Harald

thomergil, 2025-04-28:
I am not Thomas and I am not Tom. 🙂

martin2250, 2025-04-28:
To answer question number 1:

The "Start" button in the file menu makes OpenCNCPilot start sending the file. The "Start" in the top bar sends a real-time start command (~) to GRBL (see [here](https://github.com/gnea/grbl/blob/master/doc/markdown/commands.md#grbl-v11-realtime-commands))

thomergil, 2025-04-28:
Interesting. However, execution regularly pauses, which, in some cases, requires a (re-)Start in the File menu and sometimes a Start on the top bar. As a naive user, I find myself trying both Start buttons to get it going again.

martin2250, 2025-04-29:
Well an M0 in your GCode will have the same effect as a feed hold command (!). However, in the file menu there is the additional option to pause sending the file when an M0 is encountered. I think that's why you're seeing that you have to use both.