martin2250/OpenCNCPilot #203: On start, drill should first move X, Y, then Z, not the other way around.

thomergil, 2025-05-15:
When starting a print, the mill will first go down to `zsafe` and then travel to the starting point. It should do it the other way around. I know `zsafe` is supposedly ... "safe" ... but in practice choosing a large number for `zsafe` significantly slows down the milling process. So I typically choose 1mm or 2mm. When the machine has recently homed and the copper clad is in a jig, accidents happen: the mill descends to a height of 1mm and then moves laterally towards the copper clad. I have destroyed mill bits and jigs this way.

It should be an easy change to swap those two motions?

deHarro, 2025-05-15:
Hi Thomer!

As far as I understood the procedure, you should set zsafe to a literally save value.

For optimizing the probing you set the retraction height used during probing process in the settings for "probing" to the desired 1 mm or 2 mm or whatever seems fitting.

[image]

closed 2025-05-15 by thomergil

deHarro, 2025-05-15:
😮😊👍

thomergil, 2025-05-15:
You know what, I'm losing my mind!! This is not a OpenCNCPilot problem, but a `pcb2gcode` problem! I'm totally barking up the wrong tree. I'll close this and file the issue there. Sorry to distract.